kurikaesu/userspace-tablet-driver-daemon #6: HID Logs for Huion WH1409 (2048)

Dyslectric, 2021-09-23:
[buttons.txt](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/7219750/buttons.txt)
[hover.txt](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/7219751/hover.txt)
[tap-bottom-right.txt](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/7219752/tap-bottom-right.txt)
[tap-closest.txt](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/7219753/tap-closest.txt)
[tap-furthest.txt](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/7219754/tap-furthest.txt)
[tap-tilt-down.txt](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/7219755/tap-tilt-down.txt)
[tap-tilt-left.txt](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/7219756/tap-tilt-left.txt)
[tap-tilt-right.txt](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/7219757/tap-tilt-right.txt)
[tap-tilt-up.txt](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/7219758/tap-tilt-up.txt)
[tap-top-left.txt](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/7219759/tap-top-left.txt)

hover abridged for brevity

kurikaesu, 2021-09-23:
Thanks for the report. The latest commit https://github.com/kurikaesu/userspace-tablet-driver-daemon/commit/c67f4a882d2e56bf002a9bf2133cb25b9209fd2d begins the process of adding huion tablet support to the drivers.

It currently just supports the WH1409 v2 that I have but I'll be adding more devices in the next few days

kurikaesu, 2021-09-24:
I've added support for your tablet but I need you to test it with me in real-time to help fill in some missing pieces. Reach out when you can.

commit 7ff64b95 (2021-09-24): Update readme. Resolves #6
closed 2021-09-24 by kurikaesu