kurikaesu/userspace-tablet-driver-daemon #7: Support installation on Arch Linux

kurikaesu, 2021-09-24:
Since arch is a pretty popular distribution, it would be good to have it supported by the driver as a package to install without having to manually tinker with configuration etc.

Its important that the driver does not run as root.

zetagabriels, 2021-11-09:
Hello, I am an Arch Linux user. If support is added for the tablet I use, the XP-PEN Deco 03, I would be happy to help with getting this project on the AUR. I can work with you to help add support for this tablet.

kurikaesu, 2021-11-09:
Thanks for reaching out, and thanks for using nic3-14159's tool to see if the unified interface exists. From what it seems like it should be fairly straightforward to add support. I should have it done by tomorrow though I may need your help in testing it out.

zetagabriels, 2021-11-09:
Hello, the daemon is now [on the AUR.](https://aur.archlinux.org/packages/userspace-tablet-driver-daemon-git)
I will be making an AUR package for the GUI shortly.

kurikaesu, 2021-11-09:
Wow that was fast!
Are there any instructions in order to update the package or will it always pull from the latest git main branch?

kurikaesu, 2021-11-09:
That's fine. I can change the way I work with the repository by creating a development or feature branches and merge them into main when ready. It should also make it much easier for others to contribute to the code base since they won't have to be constantly rebasing.

zetagabriels, 2021-11-09:
In the AUR (and my experience), generally there are two packages per program; one that's the bleeding edge version based of `main`, suffixed with `-git`, and one that's the stable version, with no suffix. It comes down to how you want to deliver the package.

kurikaesu, 2021-11-09:
Got it, that makes a lot of sense.

I think having both a bleeding edge and a stable version is ideal since that is the same way that I am distributing the Debian packages. Stable releases and a nightly.

I'll go and create a release branch that you can use for the stable versions. We'll keep using main as the bleeding edge version.

zetagabriels, 2021-11-09:
I can also set it up to grab tags and have those be considered the releases, that way you don't have to make a whole new branch. It can target the `.tar.gz` files in the releases.

kurikaesu, 2021-11-09:
Nah a release branch works fine for me as its something I'm used to doing in my own projects.

zetagabriels, 2021-11-09:
Perfect, I'll get working on the stable AUR package later tonight.

kurikaesu, 2021-11-09:
Thanks, your help bringing this to Arch is much appreciated!

zetagabriels, 2021-11-09:
Thank you for making this package! DIGImend is neat and all, but using `xsetwacom` and the slow response times turn me off from it a bit. You are much more active with your repo.

There are two packages on the AUR now; I will maintain them!
[Stable](https://aur.archlinux.org/packages/userspace-tablet-driver-daemon/)
[Unstable](https://aur.archlinux.org/packages/userspace-tablet-driver-daemon-git/)

kurikaesu, 2021-11-11:
I can update the readme to point to your AUR packages though I'd actually prefer it if you did it so that you would show up as a contributor to the repository ^^.

Once that is done we can close this issue.

kurikaesu, 2021-11-14:
Closing this issue now as we have everything in place!

closed 2021-11-14 by kurikaesu