kurikaesu/userspace-tablet-driver-daemon #9: Test on XP-Pen 22E Pro, pen not working

Silmathoron, 2021-09-25:
Hi there, I'm trying to adapt your code to support the XP-Pen 22E Pro tablet.

I did the following:
- create ``artist_22e_pro.cpp/h`` (copy from 22r and set correct device code, 2315)
- update ``xp_pen_handler.cpp``
- update ``CMakeLists.txt``

I compiled and ran the daemon, then plugging the tablet I get the following:

```
Listening on socket ~/.local/var/run/userspace_tablet_driver_daemon.sock
xp_pen_handler initialized
huion_handler initialized
Got hotplug event
Handling XP-Pen Artist 22E Pro
Sending init key on endpont 3
Setup completed on interface 2
Set up config for device 2315
Got hotplug event
```

All buttons seem to work fine but the pen does not seem to be handled correctly (neither hovering, pressure, nor buttons lead to any action).
I know the pen works because I can use it with the XP-Pen driver (but their driver sucks, so I wanted to try with your code).

Is there something obvious I forgot that could explain why the pen is not recognized?

kurikaesu, 2021-09-25:
Thanks for the effort! Once we get this working it'd be good if you could send me a PR to merge your changes to my tree.

To the matter at hand, did you modify and copy over the xorg configuration files? They are in the config/* folders of the repository.

Specifically, you will need to add in a new record for the aliased driver:
See:
https://github.com/kurikaesu/userspace-tablet-driver-daemon/blob/main/config/usr/share/X11/xorg.conf.d/60-xp-pen.conf#L123

And the corresponding code that generates it:
https://github.com/kurikaesu/userspace-tablet-driver-daemon/blob/main/src/deco_01v2.cpp#L48-L49

You'll need to restart X for the changes to take effect.

If done correctly, `xsetwacom --list` will display the tablet name as a device.

Silmathoron, 2021-09-25:
OK, thanks for the quick reply, that's what I was missing!
I think this should definitely be on the Readme (or be more visible if it was there and I missed it) :wink: 
Should I include it in the PR or would you rather do it in a specific PR yourself?

Thanks to that the stylus is now recognized and it seems that everything more or less works but I'll need a bit of time to check everything.

In particular, I will need to test whether everything can be properly setup (area, buttons, etc).
I saw that you provided a [GUI](https://github.com/kurikaesu/userspace-tablet-driver-gui) but I did not manage to use it: ``./gradlew run`` opens a window with the tablet listed but I cannot do anything from there (right or double clicking does not open any menu so I don't know what it's supposed to do).

I'm also used to having the tablet as an additional display but by default it seems that the tablet controls the computers screen (as a non-display tablet would), is there any simple way to set the tablet display as default (either via the GUI or some other way) or do I need to set it every time via ``xsetwacom --set "XP-Pen Artist 22E Pro stylus" Area`` ?

Thanks again for your help, I'll send the PR as soon as I'm sure everything works :+1:

kurikaesu, 2021-09-25:
The GUI portion will also need some code added (Not too much) to support the device.
It will need a view:
https://github.com/kurikaesu/userspace-tablet-driver-gui/blob/main/src/main/kotlin/dev/villanueva/userland_utility/products/xppen/artist_22r_pro/Artist22rProView.kt

And a controller:
https://github.com/kurikaesu/userspace-tablet-driver-gui/blob/main/src/main/kotlin/dev/villanueva/userland_utility/products/xppen/artist_22r_pro/Artist22rProController.kt

With those set up, it has to be registered here:
https://github.com/kurikaesu/userspace-tablet-driver-gui/blob/main/src/main/kotlin/dev/villanueva/userland_utility/products/SupportedProducts.kt#L34-L92

With those in place the GUI will render up a config pane for your device when you hit "Load Configuration"

Feel free to update the Readme with any steps you think were necessary! Just make sure it is clear between what steps are general and what steps are because a new device is being supported. For example, just installing the driver as is just needs the config to be copied. Creating support for a new device needs the changes I explained above added before copying.

kurikaesu, 2021-09-25:
As far as your question about mapping the display, until I write an xorg input driver that can integrate with the rest of the userspace drivers we have to rely on xsetwacom to set the mapping as well as the stylus button bindings.

Silmathoron, 2021-09-26:
Thanks for the feedback, I'm now trying to configure the GUI (not sure whether I should open a specific issue on the other repo, let me know if you prefer that).

So far I have:
- created the ``Artist22eProView.kt`` and ``Artist22rProController.kt`` files with correct info
- added them to ``SupportedProducts.kt``

From this I now have access to the configuration window but:
- the buttons are not updated ([22E](https://www.xp-pen.com/product/57.html) only has 16 buttons, from K1 to K16, and no wheels but I still get 20 buttons and wheels on the config window even though I removed them from the controller file)
- the configuration is not applied: I can change it but it does not modify the tablet's behavior and even if I save it, after closing the GUI and opening it again, settings are back to default (which kind of makes sense as the behavior never changed).

Would you know what I missed?

kurikaesu, 2021-09-26:
I would have to look at the code changes in order to see what's different. You can submit a pull request with your changes to do so.

Silmathoron, 2021-09-26:
OK, actually I found the mistake, sorry, I just forgot to update one "r" into an "e" ^^"
Sorry for the noise.
I think everything works so I'll send both PRs ASAP (probably tomorrow) :+1:

kurikaesu, 2021-09-26:
Fantastic! I also replied to your comment on the GUI side, could you check it/test it out?

kurikaesu, 2021-09-27:
Resolved with PR #11

closed 2021-09-27 by kurikaesu