kurikaesu/userspace-tablet-driver-daemon #23: Need a tool to provide USB dump information for new devices

kurikaesu, 2021-11-09:
So far the supported devices have been devices that I personally own or from contributors who own the device themselves and were able to add the required classes and logic in order to get things supported.

In order to allow others who may not be able to write the necessary code to contribute, I am proposing writing a utility that a user can execute that will provide step by step instructions and generate tablet reports that can be used by others to implement support.

Right now when I get a new Huion tablet I uncomment these lines: https://github.com/kurikaesu/userspace-tablet-driver-daemon/blob/main/src/huion_tablet.cpp#L275-L279 and look at the dumped out bytes to see if there is any new functionality or changes to the message structure before adding the device's firmware strings to the appropriate areas: https://github.com/kurikaesu/userspace-tablet-driver-daemon/blob/main/src/huion_tablet.cpp#L115

For XP-Pen devices I do something similar by creating the class that will be dealing with the tablet and inserting the byte dump code into it and seeing the message structures.

When I have the device with me the above process takes between 15-30 mins and results in a working driver for the device at the end of it. If we have the proposed utility instead that users can run and submit, the amount of tablets that could be supported would increase much more quickly.

zetagabriels, 2021-11-09:
Would any of these tools be useful? It's not exactly the same but there could be a shell script that translates this output into something more usable by this program.

https://digimend.github.io/devel/

kurikaesu, 2021-11-09:
Those tools are currently some of what I use but they require the user to understand what they're looking at. There are also some devices where the tools cause them to stop working altogether until the device/usb controller is reset or the computer is rebooted.

What I'd like is a tool that when executed gives a list of prompts to the user:
- Touch the top left corner of tablet digitizer with stylus
- Touch the top right corner of tablet digitizer with stylus
- Enter how many buttons/shortcut keys does the device have?
- Press each key one after another at the prompts
- etc.

At the end it would spit out a text file with annotations like this (Without report descriptors since we don't need them): https://github.com/kurikaesu/DIGImend-research/blob/main/xp-pen-22r-pro.md
That would seriously make adding new devices that I don't have an absolute breeze.

zetagabriels, 2021-11-16:
I have begun a GTK client for this purpose. You can watch its progress [here](https://github.com/zetagabriels/tablet-scraper), I will leave another comment when it is ready.

kurikaesu, 2021-11-16:
Fantastic! Looking forward to your progress :)

kurikaesu, 2022-02-10:
Looks like the proposed GTK client idea has been abandoned so this issue is again open to anybody who thinks they can do it.