kurikaesu/userspace-tablet-driver-daemon #25: Test Deco 03 tablet with the driver to confirm what does and doesn't work

kurikaesu, 2021-11-10:
I don't have this tablet so if there is anybody who does, please test and report back if it works or not.

zetagabriels, 2021-11-10:
Hello! I tested using the latest version of the daemon.

The wheel does not work, and the daemon segfaults when pressing button 5 (ctrl+S). All other buttons on both the stylus and the pad work correctly.
The GUI displays it as "Unknown device".

Here is my event log and the segfault message.

[stdout.log](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/7516166/stdout.log)

```
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::reference = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
```

kurikaesu, 2021-11-10:
Thanks for the report. The GUI shows unknown device as I haven't added the necessary code into it just yet. I'll do it tonight after work.

Segfault is likely because there aren't enough buttons defined in the code. Simple fix but I need to confirm that it is the case.
If you are OK with making some code changes in order to test something, could you replace this: https://github.com/kurikaesu/userspace-tablet-driver-daemon/blob/main/src/deco_03.cpp#L125-L137

With this snippet below:
```
bool deco_03::handleTransferData(libusb_device_handle *handle, unsigned char *data, size_t dataLen) {
    std::cout << std::dec << "Got transfer of data length: " << (int)dataLen << " data: ";
    for (int i = 0; i < dataLen; ++i) {
        std::cout << std::hex << std::setfill('0')  << std::setw(2) << (int)data[i] << ":";
    }
    std::cout << std::endl;

    switch (data[0]) {
        case 0x02:
            handleDigitizerEvent(handle, data, dataLen);
            handleFrameEvent(handle, data, dataLen);
            break;

        default:
            break;
    }

    return true;
}
```

Compile and run the driver and rotate the dial left and right. It should spit out a bunch of hex code in the terminal.
Copy and paste the hex code into a reply here so I can see what the dial messages look like.

zetagabriels, 2021-11-10:
Here is the log. I rotated right, then left first. `2e` seems to correspond with a right rotation, `2d` a left rotation.

[wheel.log](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/7516324/wheel.log)

kurikaesu, 2021-11-10:
Interesting! Furthermore it looks like it sends the messages with a 0x03 prefix rather than the usual 0x02. Only other device that does something similar that I've seen so far is the deco pro S/M for the touch pad.

Thanks for the data, that should be all I need to fix everything (I think)

kurikaesu, 2021-11-11:
Could you see if commit https://github.com/kurikaesu/userspace-tablet-driver-daemon/commit/fcc204a283b532d5e7093f99112e9d55eda28719 fixes the buttons and the dials?

In the meantime I will update the GUI.

kurikaesu, 2021-11-11:
GUI has been updated to support the Deco 03

zetagabriels, 2021-11-11:
Yes! I can confirm all the buttons and dials work perfectly. The default config's zoom in/out are reversed but otherwise there are no issues. The GUI also works for rebinding all buttons; however, the stylus has three buttons on it, not two.

Thank you for adding support for this tablet!! I appreciate your work, and if there's anything you want me to do for this project (besides adding the GUI to the AUR), please let me know!

kurikaesu, 2021-11-11:
Oh interesting. Its the P05 stylus right? https://www.xp-pen.com/product/73.html
You wouldn't know that it has 3 buttons just from the looks of it.

I'll fix up the reverse dial bindings but with the same code change you made above, would you be able to send me a dump of the 3 stylus button messages? It will likely be super noisy because of the X & Y coordinates constantly being reported but I will be able to find the messages in the mess of hex anyway.

kurikaesu, 2021-11-11:
I've pushed a commit now which is just a little bit of a shot in the dark with regards to the stylus 3 button but I think it should be right. If it isn't then definitely send through the dumps if you can.

Default bindings for the dial have also been swapped.

zetagabriels, 2021-11-11:
So! I misunderstood; I assumed button 1 was the stylus tip, like how DIGImend has the stylus tip as button 0. My bad; when I first used the GUI, I thought it saved the config automatically, so I thought the button 1 config was doing nothing.

The stylus buttons 1 and 2 work with no issues, there are only two buttons.

kurikaesu, 2021-11-11:
Ok! Glad it is all working! I'm going to close this issue as I think that wraps everything up. If there's a problem you find then you can reopen it or cut a new ticket :)

closed 2021-11-11 by kurikaesu