kurikaesu/userspace-tablet-driver-daemon #30: Support for XP-Pen Artist 16 Pro

FollieHiyuki, 2021-11-12:
I have this model: https://www.xp-pen.com/product/61.html and would love to have it support.

Can you provide what I need to do to make any progress?

For context, I use Sway, so my ability is limited to only `libinput`.

FollieHiyuki, 2021-11-12:
Some basic information: 
- The ID I get from `lsusb` for the device is `28bd:090a`
- I think it is similar to the 22E Pro (but has 8 buttons instead of 16), so I will try to mimic the code there to see whether I can make something working

kurikaesu, 2021-11-12:
Thanks for looking at this! This is one of the tablets that I was considering to purchase in the future but if you are able to make it work then it is a major relief on my part.

If it is similar to the 22E pro then you can mostly copy and paste the 22e code and remove the right hand side shortcut keys.
Once cloned class is in place, you would add an instance of it here: https://github.com/kurikaesu/userspace-tablet-driver-daemon/blob/main/src/xp_pen_handler.cpp#L45

After this, compiling it and running it should be the only thing necessary to get it running. Feel free to open up a pull request early with your changes if you'd like me to look over it in case things don't work exactly.

kurikaesu, 2021-11-12:
Before closing this off we can use this issue to troubleshoot the X11 apps not working for you

FollieHiyuki, 2021-11-13:
It is a bug in XWayland, and not a driver issue. Since I already figured it out, closing.

closed 2021-11-13 by FollieHiyuki

FollieHiyuki, 2021-11-13:
@kurikaesu 1 last question though. How can I properly test tilting?

kurikaesu, 2021-11-13:
If you want to test whether the driver is sending tilt values then you can use the libinput utility like: `libinput record`. Picking the non-pad device and moving the stylus around at angles should show you tilt values as long as it is valid.

Otherwise if you are using something like Krita then you would need to test with a brush that has tilt enabled. I don't really use tilt much myself so I can't tell you which brush off the top of my head.