kurikaesu/userspace-tablet-driver-daemon #36: XP-Pen Artist 12 Pro: Unknown product 2335

allo-, 2021-11-21:
I am not sure if my device may be the untested 2nd generation (I thought not. How do I recognize it?), but the driver doesn't seem to recognize it correctly.

kurikaesu, 2021-11-21:
Looks like you have the second generation. Let me add the code required for it to be recognized

kurikaesu, 2021-11-21:
Can you pull commit https://github.com/kurikaesu/userspace-tablet-driver-daemon/commit/ad0e0cff7ced753f93481f429bb42541dfcd5464 and see if that enables your tablet?

allo-, 2021-11-21:
It is now recognized, but not working:

```
$ usr/bin/userspace_tablet_driver_daemon 
Listening on socket /home/user/.local/var/run/userspace_tablet_driver_daemon.sock
No existing config so we will be creating a new one
xp_pen_handler initialized
huion_handler initialized
Handling XP-Pen Artist 12 Pro (2nd Gen)
Setup completed on interface 0
Setup completed on interface 1
Device: XP-Pen Artist 12 Pro (2nd Gen) - Probed maxWidth: (25634) maxHeight: (14415) resolution: (2540)
Attached to interface 2
Sending init key on endpont 3
Failed to send key on interface 3 ret: -7 errno: 11
Setup completed on interface 2
Set up config for device 2335: (XP-Pen Artist 12 Pro (2nd Gen))
```

For the configuration utility I had to symlink
`xp_pen_userland -> userspace_tablet_driver_daemon`

Another question: The configuration utility does not seem to have options to configure the mapped screen area. Does the userspace driver work with multi-screen setups? I had the problem with other drivers, that they map the area of the tablet across the whole display area instead of just one screen.

kurikaesu, 2021-11-21:
Interesting. That return code of -7 means that it timed out when trying to send the init key.
Could you try running this utility to see if this device uses a different init key?: https://github.com/nic3-14159/uclogic-new-init

Looks like you are using the Python GUI if things are still referencing xp_pen_userland. Please use the Kotlin version here: https://github.com/kurikaesu/userspace-tablet-driver-gui

The configuration utility does not have the functionality to do screen mapping as it leaves all of that to X11 or Wayland.
You can see in the readme.md that I specify how to set the tablet to the specific screen including the caveat on Wayland: 
https://github.com/kurikaesu/userspace-tablet-driver-daemon/blob/main/README.md?plain=1#L84-L100

allo-, 2021-11-22:
```
Please hold the pen on the tablet surface while running this program.
Press Enter to continue
Trying init packet 0x02B002000000000000000000:
Received 02B1020000000000111D
Invalid key!

Trying init packet 0x02B004000000000000000000:
Received 02A1D14FD54CBE04111D
Key is 0x02B004000000000000000000, interface 2, endpoint 0x03
```

> You can see in the readme.md that I specify how to set the tablet to the specific screen including the caveat on Wayland:
https://github.com/kurikaesu/userspace-tablet-driver-daemon/blob/main/README.md?plain=1#L84-L100

That looks fine. I was afraid I would need to calculate the transformation matrix myself, but `map-to-output` sounds great.

allo-, 2021-11-22:
After running the utility, I had 

```
    ↳ Eee PC WMI hotkeys                      	id=11	[slave  keyboard (3)]
    ↳ SONiX USB DEVICE Keyboard               	id=12	[slave  keyboard (3)]
```

In the lsusb output instead of

```
    ↳ UGTABLET 11.6 inch PenDisplay Keyboard  	id=13	[slave  keyboard (3)]
    ↳ UGTABLET 11.6 inch PenDisplay           	id=15	[slave  keyboard (3)]
```

kurikaesu, 2021-11-22:
From the looks of the tool response we are using the correct key already for your device. Not sure why you are getting at timeout error. I'd have to debug it with an actual device to have a better idea what's going on.

The map-to-output definitely works wonders however it won't fully help if you are needing to calibrate the screen and digitizer so that the pen lines up. For that you would need to calculate the matrix by hand unfortunately as all of the calibration tools I know of don't do it correctly.

allo-, 2021-11-22:
I tried the `map-to-output` and it had no effect. (X11 / xinput / ... already provides basic support for cursor movement here).
I need to test some options as I suspect that the Nvidia-Driver with its own screen configuration may be a problem here.

kurikaesu, 2021-11-22:
Make sure when using map-to-output you are specifying the Virtual Core Pointer version of the Pen. It should have a (0) suffix to its name like:
`XP-Pen Artist 22r Pro Pen (0)`

allo-, 2021-11-22:
My IDs without the userspace driver are:

- UGTABLET 11.6 inch PenDisplay Mouse     	id=14	[slave  pointer  (2)]
- UGTABLET 11.6 inch PenDisplay Keyboard  	id=13	[slave  keyboard (3)]
- UGTABLET 11.6 inch PenDisplay           	id=15	[slave  keyboard (3)]

By the way, I see that the strange EEE PC device doesn't seem to be the tablet. Or it remained from the last run? At least I do not know which device it might be.

kurikaesu, 2021-11-22:
No idea what the EEE PC device is.
You will need to have the userspace driver running in order to get the proper IDs to give to xinput map-to-output.

Are you still getting -7 timeout errors when running the driver?

allo-, 2021-11-22:
It seems to work now. I need to check out the kotlin config utility. Is there anything else I should test?

kurikaesu, 2021-11-22:
The config utility will not be able to configure your device right now since it is missing the required classes. If you're able to look at the code and copy + paste things to make it work like this: https://github.com/kurikaesu/userspace-tablet-driver-gui/commit/1cd6df52c2a34be3cfb824b9f147275ce5133b0e
Then you should be able to get the tool to configure your tablet.

If that works then send me a PR and I'll merge it in ^^

allo-, 2021-11-22:
At the moment, the config tool does not compile. I am not sure if it's the config tool or the java in my debian testing may be broken. It segfaults in libpthread:

```
$ ./gradlew run --stacktrace

> Configure project :
Project : => no module-info.java found

> Task :run FAILED
Nov. 22, 2021 9:47:06 PM tornadofx.Stylesheet$Companion detectAndInstallUrlHandler
INFORMATION: Installing CSS url handler, since it was not picked up automatically
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f029851b0a0, pid=373367, tid=373414
#
# JRE version: OpenJDK Runtime Environment (15.0.3+3) (build 15.0.3+3-Debian-1)
# Java VM: OpenJDK 64-Bit Server VM (15.0.3+3-Debian-1, mixed mode, sharing, tiered, compressed oops, g1 gc, linux-amd64)
# Problematic frame:
# C  [libpthread.so.0+0xb0a0]  pthread_mutex_lock+0x0
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/allo/build/userspace-tablet-driver-gui/hs_err_pid373367.log
#
# If you would like to submit a bug report, please visit:
#   https://bugs.debian.org/openjdk-15
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':run'.
> Process 'command '/usr/lib/jvm/java-15-openjdk-amd64/bin/java'' finished with non-zero exit value 134

* Try:
Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Exception is:
org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':run'.
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.lambda$executeIfValid$1(ExecuteActionsTaskExecuter.java:188)
        at org.gradle.internal.Try$Failure.ifSuccessfulOrElse(Try.java:282)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeIfValid(ExecuteActionsTaskExecuter.java:186)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.execute(ExecuteActionsTaskExecuter.java:174)
        at org.gradle.api.internal.tasks.execution.CleanupStaleOutputsExecuter.execute(CleanupStaleOutputsExecuter.java:109)
        at org.gradle.api.internal.tasks.execution.FinalizePropertiesTaskExecuter.execute(FinalizePropertiesTaskExecuter.java:46)
        at org.gradle.api.internal.tasks.execution.ResolveTaskExecutionModeExecuter.execute(ResolveTaskExecutionModeExecuter.java:51)
        at org.gradle.api.internal.tasks.execution.SkipTaskWithNoActionsExecuter.execute(SkipTaskWithNoActionsExecuter.java:57)
        at org.gradle.api.internal.tasks.execution.SkipOnlyIfTaskExecuter.execute(SkipOnlyIfTaskExecuter.java:56)
        at org.gradle.api.internal.tasks.execution.CatchExceptionTaskExecuter.execute(CatchExceptionTaskExecuter.java:36)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.executeTask(EventFiringTaskExecuter.java:77)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:55)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:52)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:200)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:195)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$3.execute(DefaultBuildOperationRunner.java:75)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$3.execute(DefaultBuildOperationRunner.java:68)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:153)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:68)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.call(DefaultBuildOperationRunner.java:62)
        at org.gradle.internal.operations.DefaultBuildOperationExecutor.lambda$call$2(DefaultBuildOperationExecutor.java:79)
        at org.gradle.internal.operations.UnmanagedBuildOperationWrapper.callWithUnmanagedSupport(UnmanagedBuildOperationWrapper.java:54)
        at org.gradle.internal.operations.DefaultBuildOperationExecutor.call(DefaultBuildOperationExecutor.java:79)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter.execute(EventFiringTaskExecuter.java:52)
        at org.gradle.execution.plan.LocalTaskNodeExecutor.execute(LocalTaskNodeExecutor.java:74)
        at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:402)
        at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:389)
[... 117 lines cut ...]
        at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:68)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.call(DefaultBuildOperationRunner.java:62)
        at org.gradle.internal.operations.DefaultBuildOperationExecutor.lambda$call$2(DefaultBuildOperationExecutor.java:79)
[... cut ...]
        at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.run(DefaultPlanExecutor.java:124)
        at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:64)
        at org.gradle.internal.concurrent.ManagedExecutorImpl$1.run(ManagedExecutorImpl.java:48)
        at org.gradle.internal.concurrent.ThreadFactoryImpl$ManagedThreadRunnable.run(ThreadFactoryImpl.java:61)

* Get more help at https://help.gradle.org

BUILD FAILED in 1s
3 actionable tasks: 2 executed, 1 up-to-date

```

kurikaesu, 2021-11-22:
Its possible that the tool is incompatible with JDK-15. Could you try with JDK-11?

allo-, 2021-11-22:
```
> Configure project :
Project : => no module-info.java found

> Task :run FAILED
Nov. 22, 2021 9:53:02 NACHM. tornadofx.Stylesheet$Companion detectAndInstallUrlHandler
INFORMATION: Installing CSS url handler, since it was not picked up automatically
Is the driver not running? Could not connect to it

#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007fcf5cf280a0, pid=376982, tid=377008
#
# JRE version: OpenJDK Runtime Environment (11.0.13+8) (build 11.0.13+8-post-Debian-1)
# Java VM: OpenJDK 64-Bit Server VM (11.0.13+8-post-Debian-1, mixed mode, sharing, tiered, compressed oops, g1 gc, linux-amd64)
# Problematic frame:
# C  [libpthread.so.0+0xb0a0]  pthread_mutex_lock+0x0
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/allo/build/userspace-tablet-driver-gui/hs_err_pid376982.log
#
# If you would like to submit a bug report, please visit:
#   https://bugs.debian.org/openjdk-11
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':run'.
> Process 'command '/usr/lib/jvm/java-11-openjdk-amd64/bin/java'' finished with non-zero exit value 134

* Try:
Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Exception is:

org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':run'.
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.lambda$executeIfValid$1(ExecuteActionsTaskExecuter.java:188)
        at org.gradle.internal.Try$Failure.ifSuccessfulOrElse(Try.java:282)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeIfValid(ExecuteActionsTaskExecuter.java:186)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.execute(ExecuteActionsTaskExecuter.java:174)
        at org.gradle.api.internal.tasks.execution.CleanupStaleOutputsExecuter.execute(CleanupStaleOutputsExecuter.java:109)
        at org.gradle.api.internal.tasks.execution.FinalizePropertiesTaskExecuter.execute(FinalizePropertiesTaskExecuter.java:46)
        at org.gradle.api.internal.tasks.execution.ResolveTaskExecutionModeExecuter.execute(ResolveTaskExecutionModeExecuter.java:51)
        at org.gradle.api.internal.tasks.execution.SkipTaskWithNoActionsExecuter.execute(SkipTaskWithNoActionsExecuter.java:57)
        at org.gradle.api.internal.tasks.execution.SkipOnlyIfTaskExecuter.execute(SkipOnlyIfTaskExecuter.java:56)
        at org.gradle.api.internal.tasks.execution.CatchExceptionTaskExecuter.execute(CatchExceptionTaskExecuter.java:36)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.executeTask(EventFiringTaskExecuter.java:77)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:55)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:52)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:200)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:195)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$3.execute(DefaultBuildOperationRunner.java:75)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$3.execute(DefaultBuildOperationRunner.java:68)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:153)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:68)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.call(DefaultBuildOperationRunner.java:62)
        at org.gradle.internal.operations.DefaultBuildOperationExecutor.lambda$call$2(DefaultBuildOperationExecutor.java:79)
        at org.gradle.internal.operations.UnmanagedBuildOperationWrapper.callWithUnmanagedSupport(UnmanagedBuildOperationWrapper.java:54)
        at org.gradle.internal.operations.DefaultBuildOperationExecutor.call(DefaultBuildOperationExecutor.java:79)
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter.execute(EventFiringTaskExecuter.java:52)
        at org.gradle.execution.plan.LocalTaskNodeExecutor.execute(LocalTaskNodeExecutor.java:74)
        at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:402)
        at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:389)
        at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.execute(DefaultTaskExecutionGraph.java:382)
[... 110 lines cut ...]
        at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:52)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:200)
[... cut ...]
        at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.executeNextNode(DefaultPlanExecutor.java:182)
        at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.run(DefaultPlanExecutor.java:124)
        at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:64)
        at org.gradle.internal.concurrent.ManagedExecutorImpl$1.run(ManagedExecutorImpl.java:48)
        at org.gradle.internal.concurrent.ThreadFactoryImpl$ManagedThreadRunnable.run(ThreadFactoryImpl.java:61)
```

kurikaesu, 2021-11-23:
That is very weird. What distribution and version are you using? I may need to replicate the issue locally.

allo-, 2021-11-23:
Debian testing. So think *may* be broken by Debian. But I think java should work at the moment. Maybe I should debug with strace and gdb? I am not sure how much native code kotlin is using and if the segfault happens in the jvm (which would be a java bug?) or in native code.

kurikaesu, 2021-11-23:
Hmmn that's a good question. I can definitely just give it a go on my side using something like a VM to see what is going on.

kurikaesu, 2021-11-24:
Just tested the GUI on a fresh install of Debian Testing and it works for me using OpenJDK-11

[image]

allo-, 2021-11-24:
I am not sure about it. But the driver itself works fine when configured using the python utility (and changing the device ID in the artist 12 pro class).

kurikaesu, 2021-11-24:
Then I don't know whats going on since it isn't reproducible on my side. If the python utility works then that's OK though I won't be updating the utility myself. Feel free to send me a PR for the python utility to add your device.

allo-, 2021-11-24:
I guess I rather get the Java program to work when I have time, but until then I can already paint. :)

allo-, 2021-11-24:
So I guess this bug can be closed as the driver works with the tablet?

closed 2021-11-24 by allo-

kurikaesu, 2021-11-24:
Yes. I will add the code required to the kotlin gui today