kurikaesu/userspace-tablet-driver-daemon #39: XP-Pen 15.6 Pro

davix3f, 2021-12-13:
DEVICE ID:  28bd:090d
PRODUCT NAME:  XP-Pen Artist 15.6 Pro
PRODUCT PAGE:   https://www.xp-pen.it/product/409.html

The device correctly works with the userspace driver and the screen mapping can be changed with xinput without any problem, however it is not detected by the config GUI so I couldn't change the button mappings as I couldn't find the JSON config file as well.

kurikaesu, 2021-12-14:
Thanks for the report, I'd almost thought I had implemented official support for this tablet but looking at my readme file its clear that I hadn't!

I'll add the IDs and update the config utility to allow it to work.

kurikaesu, 2021-12-15:
I've added commit https://github.com/kurikaesu/userspace-tablet-driver-gui/commit/50da1107f23001e3c75f7610ec4cac8c4ec6dc8d to the GUI. Please pull the latest commits from both the daemon and GUI repositories and let me know if everything is working on your end!

davix3f, 2021-12-15:
Git-pulled and rebuilt the gui, the tablet still doesn't show up in the window.
When I start the driver, the tablet is labeled as "Unknown product 2317":

```
davide@arch >> userspace_tablet_driver_daemon
Listening on socket /home/davide/.local/var/run/userspace_tablet_driver_daemon.sock
xp_pen_handler initialized
huion_handler initialized
Unknown product 2317
Got new socket connection
```
I noticed that in 60-xp-pen.conf the section for the 15.6 Pro is missing, maybe that's the issue?

If that can help, that is the output when I run the GUI

```
davide@arch >> ./gradlew run

> Configure project :
Project : => no module-info.java found

> Task :run

(java:28781): Gtk-WARNING **: 04:22:19.179: Theme parsing error: gtk.css:73:46: The style property GtkScrolledWindow:scrollbars-within-bevel is deprecated and shouldn't be used anymore. It will be removed in a future version
dic 15, 2021 4:22:19 AM tornadofx.Stylesheet$Companion detectAndInstallUrlHandler
INFO: Installing CSS url handler, since it was not picked up automatically
````

kurikaesu, 2021-12-15:
Ah, I missed adding the 15.6 pro details to the Xorg config. I'll double check what's going on but I could have missed a few more things. Apologies!

kurikaesu, 2021-12-15:
Oh actually after checking my git client, I hadn't actually pushed the code yet to the daemon repository. I've done the push with the commit: https://github.com/kurikaesu/userspace-tablet-driver-daemon/commit/18009b74422cd68c343d1db2eab918a30f563c57

Please pull that and rebuild. It should recognise your tablet this time around.

davix3f, 2021-12-15:
Tablet is still labeled as "Unknown Product 2317". I rebuilt the driver directly from the git repo; no errors in compiling, but executing the driver makes the tablet unable to even detect the pen and doesn't show up in `xinput`.
Also, executing the git driver as normal user results in a lot of binary chars printed in the output for some reason; their amount increases if I hover the pen above the tablet.

```
>> userspace_tablet_driver_daemon
Listening on socket /home/davide/.local/var/run/userspace_tablet_driver_daemon.sock
xp_pen_handler initialized
huion_handler initialized
Setup completed on interface 0
Setup completed on interface 1
Device: Generic XP-Pen Device - Probed maxWidth: (34419) maxHeight: (19461) resolution: (2540)
Could not create uinput pen
Could not create uinput pad
Attached to interface 2
Sending init key on endpont 3
Setup completed on interface 2
Set up config for device 2317: (Generic XP-Pen Device)
Unknown product 2317
��a�
    @��a�
         ▒��a�
              ��a�
                  ��a�
                      ▒��a�
                           �a�
```

Running the driver as sudo doesn't result in such behaviour and the tablet works again, but the gui program still won't detect it - and can't connect to the driver. Running the gui program as sudo basically makes it crash

Driver as sudo:
```
>> sudo userspace_tablet_driver_daemon
[sudo] password di davide:         
Listening on socket /root/.local/var/run/userspace_tablet_driver_daemon.sock
No existing config so we will be creating a new one
xp_pen_handler initialized
huion_handler initialized
Setup completed on interface 0
Setup completed on interface 1
Device: Generic XP-Pen Device - Probed maxWidth: (34419) maxHeight: (19461) resolution: (2540)
Attached to interface 2
Sending init key on endpont 3
Setup completed on interface 2
Set up config for device 2317: (Generic XP-Pen Device)
Unknown product 2317
```
GUI as sudo:
```
>> sudo ./gradlew run

> Configure project :
Project : => no module-info.java found

> Task :run
dic 15, 2021 2:14:13 PM tornadofx.Stylesheet$Companion detectAndInstallUrlHandler
INFORMAZIONI: Installing CSS url handler, since it was not picked up automatically
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f68a4a06424, pid=7558, tid=7589
#
# JRE version: OpenJDK Runtime Environment (17.0.1+12) (build 17.0.1+12)
# Java VM: OpenJDK 64-Bit Server VM (17.0.1+12, mixed mode, tiered, compressed oops, compressed class ptrs, g1 gc, linux-amd64)
# Problematic frame:
# C  [libpthread.so.0+0xb424]  pthread_mutex_lock+0x4
#
# Core dump will be written. Default location: Core dumps may be processed with "/usr/lib/systemd/systemd-coredump %P %u %g %s %t %c %h" (or dumping to /home/davide/userspace-tablet-driver-gui/core.7558)
#
# An error report file with more information is saved as:
# /home/davide/userspace-tablet-driver-gui/hs_err_pid7558.log
#
# If you would like to submit a bug report, please visit:
#   https://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

> Task :run FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':run'.
> Process 'command '/usr/lib/jvm/java-17-openjdk/bin/java'' finished with non-zero exit value 134

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 8s
3 actionable tasks: 2 executed, 1 up-to-date
```

This is the error log from running gui as sudo: https://pastebin.com/tQiYsW5X

kurikaesu, 2021-12-15:
I missed adding the 15.6 pro to the list of handled models in xp_pen_handler.cpp. I've added it in now.
The driver does not need sudo to work and you really don't want it to be running under sudo as it will store all the config files under /root/.

Does the GUI segfault when run not under sudo/root?

davix3f, 2021-12-15:
Now the GUI works (running it as a normal user, but I don't mind), detects the device and can change its config.
However, if I don't run the driver as sudo, I will get this output:

```
>> userspace_tablet_driver_daemon
Listening on socket /home/davide/.local/var/run/userspace_tablet_driver_daemon.sock
xp_pen_handler initialized
huion_handler initialized
Handling XP-Pen Artist 15.6 Pro
Setup completed on interface 0
Setup completed on interface 1
Device: XP-Pen Artist 15.6 Pro - Probed maxWidth: (34419) maxHeight: (19461) resolution: (2540)
Could not create uinput pen
Could not create uinput pad
Attached to interface 2
Sending init key on endpont 3
j�a"�   @j�a5�  ▒j�aD�  Kj�aL�  �Qj�aT� �j�a]�  ▒����j�ad�      j�al�   Setup completed on interface 2
Set up config for device 2317: (XP-Pen Artist 15.6 Pro)
j�a$�   ▒j�aB�  Kj�aJ�  j�aZ�   j�ab�   ▒����j�ai�      j�ap�
```

All those strange chars are printed everytime the pen is hovered on the tablet, and I can't use it as a pointing device. This didn't happen with the AUR (non-git) version of the driver. I believe that the lines
```
Could not create uinput pen
Could not create uinput pad
```
are telling something about permissions or something like that, because the driver will work fine if I run it as sudo. Do I have to add my user to some group maybe?

davix3f, 2021-12-15:
```
davide@arch >> groups
wireshark wheel davide
```

kurikaesu, 2021-12-15:
After adding the user you will likely need to do a:
`sudo udevadm trigger` in order for the permissions on /dev/uinput to be updated

davix3f, 2021-12-15:
Yes, running that command fixed that issue. Now the driver works fine even when not running as root. Maybe this group check could go into the make-install process?
The thing is, the buttons don't work. Pressing tablet or pen buttons has no effect whatsoever.

I don't know if this should be a separate issue, but the GUI behaves really strange. I cannot remap buttons at all and it does not mantain the configuration, emptying every button everytime I restart the GUI.

Where is the config file located?

kurikaesu, 2021-12-15:
Config file is located in $HOME/.local/share/userspace_tablet_driver_daemon/driver.cfg
If you ran the daemon driver with sudo then it is possible that the ownership of the config is "root" and you are unable to modify it to update bindings. If this is the case then you just need to change the ownership of the file.

davix3f, 2021-12-15:
I chmod777'd both the user and root driver.cfg, but nothing really changed. I cannot edit what buttons do, be it tablet or pen buttons. I took a screencast of how the gui behaves https://cdn.bunkr.is/2021-12-15-20-47-00-rfjsS41Q.mkv

kurikaesu, 2021-12-16:
From the looks of it, the code that is handling accepting inputs is hanging or not resolving correctly. If you pick any of the non-stylus buttons and press a key to add it, does it change it correctly? There is no Artist 15.6 pro specific code in the GUI other than adding the correct IDs and list of buttons.

I'll test things locally on my 13.3 that I based the code from to see if something has gone weird when I get a moment.

kurikaesu, 2022-02-12:
I've tested this with both my Artist 24 and Innovator 16 and I don't see the issue. I'll close this off as I can't reproduce the problem. Feel free to re-open this issue if you are able to get some more debugging happening.

closed 2022-02-12 by kurikaesu