kurikaesu/userspace-tablet-driver-daemon #45: Star G960S - Configuration help

bferguson3, 2022-01-24:
The configuration file is excessively obtuse and has no information at all in it. Spec changes aside its just a bunch of numbers, meaning the GUI is the only viable way to configure your tablet buttons.

Except the GUI only compiles on AMD architectures. I'm on ARM64. What do I do?

kurikaesu, 2022-01-25:
Yes the configuration file is not really designed to be hand modified. It is just full of device IDs, keycode and wheel scancodes that can be a little tricky to modify.

I do not own an ARM64 device so I can't devise a suitable alternative there yet. Possible solutions are to create a non-Java based configuration utility and use something like GTK or Qt directly.

bferguson3, 2022-01-25:
The issue appears to be exclusively with openjfx/javafx. I am able to
install libjavafx fine, so maybe you can optionally dynamically link it.

In the meanwhile, I would really appreciate some information on how to
change the bindings. I don't care if it might change in the future.

kurikaesu, 2022-01-25:
Here is a snippet of the config that is pretty printed:

[PrettyPrinted.txt](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/7936734/PrettyPrinted.txt)

Under device configurations is a map that contains the different vendor IDs in decimal format. The entry `10429` is in hex `0x28BD` which corresponds to the XP-Pen vendor: https://github.com/kurikaesu/userspace-tablet-driver-daemon/blob/main/src/xp_pen_handler.cpp#L64

Under this dictionary entry will be the product ID in decimal format. For G960S I have no idea what the product ID is because I haven't had one in my possession before. An example I can give is the Deco Pro Small that has the product ID in hex of 0x0909: https://github.com/kurikaesu/userspace-tablet-driver-daemon/blob/main/src/deco_pro_small.cpp#L24 which converts to decimal to 2313.

Searching for that product ID, you get a mapping dictionary that gives buttons, dials and stylus_buttons.

For buttons, these correspond to the scancodes that the actual buttons produce: https://github.com/kurikaesu/userspace-tablet-driver-daemon/blob/main/src/deco_pro.cpp#L25-L27

The BTN_0 is defined under the linux headers: https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h#L343-L416

You would translate the hex code numbers into decimal to get the correct button map listed in the config file.

Once you figure out which one it is, underneath the desired button would be a simple `1` entry which translates to an enum used in my driver that points to an ordered list of input scan codes. These scan codes are your keyboard layout specific. EN_US QWERTY layout would have the key `B` as 48: https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h#L123

Any modifiers that you need to have happen, for example a shortcut for `CTRL + '+'` needs to have the modifier key first in the list as the driver will send the keys in order. Doing a `'+' + CTRL` is likely an undesired operation in this case and will result in the actual application shortcut not executing correctly.

Hope that helps!

bferguson3, 2022-01-25:
This is excellent, thank you very much. I will see about writing a linux
based configuration program.

kurikaesu, 2022-02-12:
I'm closing this issue as there is nothing further to do. Feel free to re-open if you need to.

closed 2022-02-12 by kurikaesu