kurikaesu/userspace-tablet-driver-daemon #46: Running on wayland

albfan, 2022-01-25:
Hi

I just see comments on README about wayland. I'm using it:

```
echo $XDG_SESSION_TYPE 
wayland
```

After review what daemon this installs:

```
$ sudo make install
[sudo] password for alberto: 
[100%] Built target userspace_tablet_driver_daemon
Install the project...
-- Install configuration: ""
-- Up-to-date: /usr/bin/userspace_tablet_driver_daemon
-- Up-to-date: /etc
-- Up-to-date: /etc/udev
-- Up-to-date: /etc/udev/rules.d
-- Up-to-date: /etc/udev/rules.d/70-huion.rules
-- Up-to-date: /etc/udev/rules.d/70-xp-pen.rules
-- Up-to-date: /etc/udev/rules.d/70-uinput-plugdev.rules
-- Up-to-date: /usr
-- Up-to-date: /usr/share
-- Up-to-date: /usr/share/X11
-- Up-to-date: /usr/share/X11/xorg.conf.d
-- Up-to-date: /usr/share/X11/xorg.conf.d/60-xp-pen.conf
-- Up-to-date: /usr/share/X11/xorg.conf.d/60-huion.conf
```

it just writes to: `~/.local/var/run/userspace_tablet_driver_daemon.sock`

I started it:

$ userspace_tablet_driver_daemon 
Listening on socket /home/alberto/.local/var/run/userspace_tablet_driver_daemon.sock
No existing config so we will be creating a new one
xp_pen_handler initialized
huion_handler initialized
Handling Huion tablet
Got firmware HUION_T200_210309
Resolved device name to Huion KD100 mini Keydial
Attached to interface 0
Setup completed on interface 0
Setup completed on interface 1
Setup completed on interface 2
Set up config for device 512: (Unknown Huion device)

and it was able to read my config (Huion KD100):

[image]

[image]

And modify keys.

So there's only lack of how to expose those keys in wayland (because yes, it only outputs garbage)

albfan, 2022-01-25:
It's suppose the using libei, or a tool that uses lbei: https://gitlab.freedesktop.org/libinput/libei/

something like uinput would also work, but it's lower level

@kurikaesu any tips of where to implement this? I'm tempted to detect wayland using "XDG_SESSION_TYPE" env var and use this library to send keys

kurikaesu, 2022-01-25:
Interesting. I've tested my tablets under wayland and the express/shortcut keys appear to work correctly. Granted, I have not done this anytime recently. I wonder if it is because the KD100 is not being identified as a keyboard device. With xorg/X11 the configuration files force the device to be handled by libinput.

albfan, 2022-01-26:
I own a HS611 and default customization provided by GNOME mostly works (it just miss some side bar keys and upper scroll)

[image]

for KD100 

[image]

looks it behaves as no driver mode when just connected, 

https://support.huion.com/en/support/solutions/articles/44002245976-huion-mini-keydial-kd100-faqs

although it doesn't match for me. It exposes: 
k1 Ctrl+Z 
k2 Ctrl+Shift+Z
k5 Ctrl+Space
k6 F5

rest of keys and dial are empty.

I need to clear any setup I made `rm ~/.local/share/userspace_tablet_driver_daemon/driver.cfg` or no key works for me.

I understand from lsusb:

    Bus 001 Device 005: ID 256c:006d

I need to rework something on `src/huion_handler.cpp` but not sure where 

and using your userspace server it outputs

albfan, 2022-01-26:
hmm looks like it is behaving as keystrokes for GNOME tablet:

[image]

it only exposes 8 keys, while this device has 16 square buttons + two large buttons + dial + 1 mouse button

still your userspace server + wayland looks interesting to implement to me

albfan, 2022-01-26:
Finally my approach was to expose how you detect real device through wireless usb dongle (firmware) here:

https://github.com/linuxwacom/libwacom/issues/469

I think expose device through a common library api allows to deal with devices easily.

what do you think? Still able to collaborate here to add wayland support if you need. Reopen if you want to see it implemented

closed 2022-01-26 by albfan