kurikaesu/userspace-tablet-driver-daemon #54: Configuration file is overwritten with default values

jbigler, 2022-05-01:
I haven't been able to determine what the cause is, but many times (though not always) when starting the daemon it overwrites the `~/.local/share/userspace_tablet_driver_daemon/driver.cfg` with default values. If I stop the daemon, restore the driver and start the daemon manually again it usually works fine. 
Right now I have it launching from `.xprofile `.

kurikaesu, 2022-05-01:
Hmmn that is very weird. Is there any way you can set up the launch parameters so that it will pipe the console output to a log file somewhere? I'm curious to know why it is deciding to reset to default.

jbigler, 2022-05-08:
Since I've started logging the console output I haven't experience the problem. Nice little Heisenbug. I'll let you know if it occurs under observation.

jbigler, 2022-05-15:
I will attach a log file I created. You'll notice sometimes the daemon says it could not claim the interface or device, but that doesn't seem to delete the configuration. On line 240 of the log file it just pops up a message that there is no existing config file and creates a new one.
[tablet-driver.log](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/8695094/tablet-driver.log)

kurikaesu, 2022-05-16:
Hmm, thanks for the log. I wonder if the reason why the config is being recreated is because the existing config is broken or half written and can't be deserialised correctly.

I'll have to look into this deeper when I can get my Linux computer hooked up to a monitor and keyboard. Might be able to do that tomorrow evening.

kurikaesu, 2022-05-23:
I've made a small change to flush the config file when writing and also emit a message to know that it has done so. If you could pull the latest code and run it while logging the output again, we can see if it is indeed being able to write out the whole file or not.

jbigler, 2022-06-26:
[userspace.log](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/8986748/userspace.log)
Sorry for the delay, I recently moved and it took a little while to get my computer set up again. Here is an attached log file from the updated build with several instances of creating a new configuration file instead of using the existing file.
When I