kurikaesu/userspace-tablet-driver-daemon #58: XP-Pen Artist 16 Pro support

ThatOneCalculator, 2022-09-09:


kurikaesu, 2022-09-09:
Hello, is there an issue you are experiencing?

ThatOneCalculator, 2022-09-09:
It doesn't detect it at all, also it's not on the compatability list yet

kurikaesu, 2022-09-09:
We have it implemented here: https://github.com/kurikaesu/userspace-tablet-driver-daemon/blob/main/src/artist_16_pro.cpp

I suspect your version of the 16 Pro is a newer revision that is different to the one we already have.

ThatOneCalculator, 2022-09-09:
How would I find the device ID?

kurikaesu, 2022-09-09:
You can use `lspci` https://man7.org/linux/man-pages/man8/lspci.8.html 
or `lsusb` : https://linux.die.net/man/8/lsusb

It will show the device ID like:
27AC:4325

ThatOneCalculator, 2022-09-09:
`Bus 001 Device 005: ID 28bd:094b XP-Pen Artist Pro 16`

kurikaesu, 2022-09-09:
Yup, looks like a new model. The current model we support is 28bd:090a

I'll make some code changes and I'll reply here when I've pushed them so you can test them out 😄

kurikaesu, 2022-09-10:
Apologies for the wait, I had to get a spare computer set up with Linux.

Please pull this commit and test to see if it works: https://github.com/kurikaesu/userspace-tablet-driver-daemon/commit/9018646944351363f36a77fa0acb49c484d1c5bc

The GUI is not set up yet. Once you can confirm the tablet works with the above changes then I'll make the changes to the GUI :)

ThatOneCalculator, 2022-09-10:
Still not getting anything, but I want to make sure I'm doing it right:

- `yay -S userspace-tablet-driver-daemon-git --overwrite "*"`
- `sudo udevadm trigger`
- Reboot

ThatOneCalculator, 2022-09-10:
It doesn't really work that well.

What happens is the cursor is locked to the right-hand side of the screen and can't move horizontally. Vertical movement seems to work fine however. I'm on Wayland with Hyprland.

```
$ userspace_tablet_driver_daemon
Listening on socket /home/kainoa/.local/var/run/userspace_tablet_driver_daemon.sock
xp_pen_handler initialized
huion_handler initialized
Saved updated configuration file
Setup completed on interface 0
Setup completed on interface 1
Device: Generic XP-Pen Device - Probed maxWidth: (2678) maxHeight: (38359) resolution: (5080)
Attached to interface 2
Sending init key on endpont 3
Setup completed on interface 2
Set up config for device 2379: (Generic XP-Pen Device)
Unknown product 2379
^CCaught SIGINT
Shutting down
Saved updated configuration file
````

kurikaesu, 2022-09-10:
The maxWidth that the generic driver is detecting is not right. It looks like structure of USB messages that your tablet is sending is different to what has been seen up until now.

I seem to have forgotten to add the new handler to the XP Pen supported device registry so I'll make another commit to fix that. I'll also have it spit out debugging info. Once I have the commit pushed, I'll need you to help me by pasting in all the logging output it will start dumping

ThatOneCalculator, 2022-09-10:
Got it, ping me when it's ready

kurikaesu, 2022-09-10:
Pushed the commit https://github.com/kurikaesu/userspace-tablet-driver-daemon/commit/62b1f91db55257194e04d0afb711b90cc58f0175

Please pull and try it out.
It will start spitting out lines like `Got transfer of data length: ....` and a whole bunch of hexadecimal when you you move your stylus across the tablet. Could you copy and paste in that data?

Specifically I'll need data from where the stylus is in the top left, top right, bottom left and bottom right. This way I can find out where the X and Y coordinates are being stored in the USB packet.

ThatOneCalculator, 2022-09-10:
Odd, I still get the old message.

```
Listening on socket /home/kainoa/.local/var/run/userspace_tablet_driver_daemon.sock
xp_pen_handler initialized
huion_handler initialized
Saved updated configuration file
Setup completed on interface 0
Setup completed on interface 1
Device: Generic XP-Pen Device - Probed maxWidth: (2678) maxHeight: (38359) resolution: (5080)
Attached to interface 2
Sending init key on endpont 3
Setup completed on interface 2
Set up config for device 2379: (Generic XP-Pen Device)
Unknown product 2379
^CCaught SIGINT
Shutting down
Saved updated configuration file
```

ThatOneCalculator, 2022-09-10:
Should I try uninstalling the AUR and build from source instead?

kurikaesu, 2022-09-10:
Yes I would build from source for now

ThatOneCalculator, 2022-09-10:
Built from source and got the exact same thing... maybe it's not pointing to the right binary?

```
$ whereis userspace_tablet_driver_daemon 
userspace_tablet_driver_daemon: /usr/bin/userspace_tablet_driver_daemon
```

ThatOneCalculator, 2022-09-10:
Actually, running `./userspace_tablet_driver_daemon` in the build dir gives the exact same thing :(

kurikaesu, 2022-09-10:
Can you check what is on line 50 of the file `src/xp_pen_handler.cpp` in your local source code?

It should have: `    addHandler(new artist_pro_16());`

ThatOneCalculator, 2022-09-10:
Apologies for the delay, it does have it

[image]

kurikaesu, 2022-09-10:
Thanks, looks like I missed one more thing. Apologies. Before I go and make this change though, could you open up `/src/xp_pen_unified_device.cpp` and on line 36, change the 12 to 13?

This is the line in question: https://github.com/kurikaesu/userspace-tablet-driver-daemon/blob/main/src/xp_pen_unified_device.cpp#L36

Run the driver and see if the maxWidth value changes to a much larger number than 2678 ?

ThatOneCalculator, 2022-09-10:
Apologies for the delay in my response. Changing the `12` to `13` seems to have done the trick, it now works properly!

```
Listening on socket /home/kainoa/.local/var/run/userspace_tablet_driver_daemon.sock
xp_pen_handler initialized
huion_handler initialized
Saved updated configuration file
Setup completed on interface 0
Setup completed on interface 1
Device: Generic XP-Pen Device - Probed maxWidth: (68214) maxHeight: (38359) resolution: (5080)
Attached to interface 2
Sending init key on endpont 3
Setup completed on interface 2
Set up config for device 2379: (Generic XP-Pen Device)
Unknown product 2379
^CCaught SIGINT
Shutting down
Saved updated configuration file
```

kurikaesu, 2022-09-10:
68214 sounds right to me. It isn't screen resolution (1920x1080) but rather the tablet sensor resolution.

The Pro 16 has a width of approximately 13.4 inches.
68214 ÷ 13.4 = roughly 5,090.

According to the product detail page: https://www.xp-pen.com/product/977.html the device has a resolution of 5080 LPI (lines per inch) which is pretty close to the 5090 I have calculated above.

I'll go finish off the fixes!

ThatOneCalculator, 2022-09-10:
Wonderful, thank you for the help! 

On a side note, do I need both the GUI and the Daemon or do I choose just one of them?

kurikaesu, 2022-09-10:
Please pull the latest commit: https://github.com/kurikaesu/userspace-tablet-driver-daemon/commit/91e49f74d6a2352d08acd8d989fbf7fd734c5882

You'll need both the Daemon and the GUI unless you are fine with the default shortcut bindings.

The GUI is just so you can customize what the express/shortcut keys on the tablet and the buttons on the stylus do.

Let me know if the latest commit has everything working for you!
Please check the pen pressure by the way!

kurikaesu, 2022-09-10:
Once everything is working on the Daemon side, I'll update the GUI

ThatOneCalculator, 2022-09-10:
Pen pressure works, I'm also now getting the outputs like `Got transfer of data length: 12 data: 02:a0:5f:95:45:4c:00:00:00:0d:00:00:`

Funnily enough on the scroll wheel, zooming out works but not zooming in!

[image]

kurikaesu, 2022-09-10:
Zooming is totally dependent on your keyboard layout. This is where the GUI comes into play. I'll go clean up the debug output and push a final commit here. Then I'll go do the GUI and post back here.

kurikaesu, 2022-09-10:
I've pushed changes to both repositories. Could you test the GUI out as well?

ThatOneCalculator, 2022-09-10:
I'm having trouble building the GUI.

```
> Configure project :
Project : => no module-info.java found

> Task :compileKotlin FAILED
e: /tmp/userspace-tablet-driver-gui/src/main/kotlin/dev/villanueva/userland_utility/products/xppen/artist_pro_16/ArtistPro16View.kt: (6, 31): Unresolved reference: Innovator16Controller
e: /tmp/userspace-tablet-driver-gui/src/main/kotlin/dev/villanueva/userland_utility/products/xppen/artist_pro_16/ArtistPro16View.kt: (6, 56): Property delegate must have a 'getValue(ArtistPro16View, KProperty<*>)' method. None of the following functions is suitable: 
public abstract operator fun getValue(thisRef: Component, property: KProperty<*>): ??? defined in kotlin.properties.ReadOnlyProperty

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileKotlin'.
> Compilation error. See log for more details

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 12s
1 actionable task: 1 executed

> Configure project :
Project : => no module-info.java found

> Task :compileKotlin FAILED
e: /tmp/userspace-tablet-driver-gui/src/main/kotlin/dev/villanueva/userland_utility/products/xppen/artist_pro_16/ArtistPro16View.kt: (6, 31): Unresolved reference: Innovator16Controller
e: /tmp/userspace-tablet-driver-gui/src/main/kotlin/dev/villanueva/userland_utility/products/xppen/artist_pro_16/ArtistPro16View.kt: (6, 56): Property delegate must have a 'getValue(ArtistPro16View, KProperty<*>)' method. None of the following functions is suitable: 
public abstract operator fun getValue(thisRef: Component, property: KProperty<*>): ??? defined in kotlin.properties.ReadOnlyProperty

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileKotlin'.
> Compilation error. See log for more details

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 2s
1 actionable task: 1 executed
```

ThatOneCalculator, 2022-09-10:
https://github.com/kurikaesu/userspace-tablet-driver-gui/pull/12 lets it compile, but there's no options on the GUI.

[image]

ThatOneCalculator, 2022-09-10:
https://github.com/kurikaesu/userspace-tablet-driver-gui/blob/37ec66b45e44476ad52dfaabb2ce4a9f0732d454/src/main/kotlin/dev/villanueva/userland_utility/products/xppen/artist_pro_16/ArtistPro16Controller.kt#L18-L19

        mapItems.add(MappableItem(MappableItemType.Dial, "Dial -> Left", DriverCodeIDs.REL_WHEEL.code, -1))
        mapItems.add(MappableItem(MappableItemType.Dial, "Dial -> Right", DriverCodeIDs.REL_WHEEL.code, 1))
        

Interestingly, those are there so I'm not sure why it doesn't show up in the GUI :thinking:

ThatOneCalculator, 2022-09-11:
Also, the zooming problem was fixable by rebinding it in Krita. It sends <kbd>Ctrl</kbd>+<kbd>=</kbd> instead of <kbd>Ctrl</kbd>+<kbd>+</kbd>.

kurikaesu, 2022-09-11:
🤦 Its because of this: https://github.com/kurikaesu/userspace-tablet-driver-gui/blob/37ec66b45e44476ad52dfaabb2ce4a9f0732d454/src/main/kotlin/dev/villanueva/userland_utility/products/xppen/artist_pro_16/ArtistPro16View.kt#L6

Easy fix! (Hopefully that is all that is wrong)

On US layout keyboards + and = are the same key, and actually pressing the + requires `CTRL + SHIFT + =`
Definitely not something I can always take care of hence the GUI to allow you to rebind stuff.

ThatOneCalculator, 2022-09-11:
After testing I can safely say that this can be closed, as #59 / #60 are their own thing. Thank you so much for writing in support for my tablet!

[image]

closed 2022-09-11 by ThatOneCalculator