kurikaesu/userspace-tablet-driver-daemon #59: Dual roller support

ThatOneCalculator, 2022-09-10:
Inner roller settings are not present in the code. From the official drivers:

Outer roller:
[image]

Inner roller (touch):
[image]

kurikaesu, 2022-09-11:
Yeah I need to implement the inner roller on my Innovator 16 as well. Just haven't gotten around to doing it, mostly because I don't use it 😓

Holmes5, 2023-01-26:
How hard would it be for someone else to implement? What would be the steps, if it's not too hard to explain? Is it plausible that I, a data scientist who mostly programs in Python and knows C++ syntax, but has no real C++ experience and certainly no experience writing drivers, could figure it out?

kurikaesu, 2023-01-29:
I don't see why you wouldn't be able to do it.

The inner roller for this device is probably similar to the other dual roller devices already implemented prior. Look at how the Deco Pro Medium/Small devices are done or how the Artist Pro 24 is implemented.

You will need to dump the USB packets in order to determine whether or not the bit position of the roller messages are in the same location as it is for the other devices. Refer to this for a good reference: https://github.com/kurikaesu/DIGImend-research/blob/main/xp-pen-22r-pro.md

You can modify the source code and have it dump out the actual USB protocol messages by copying and uncommenting these lines to the appropriate driver code: https://github.com/kurikaesu/userspace-tablet-driver-daemon/blob/main/src/huion_tablet.cpp#L302-L306 

Hope that helps!

If you figure it out, feel free to send me a PR!