kurikaesu/userspace-tablet-driver-daemon #60: Eraser support

ThatOneCalculator, 2022-09-10:
There's nothing in the code with support for erasers. On the XP-Pen Artist Pro 16 drivers, you can flip your pen around to "draw" with the eraser on the top, which presses <kbd>E</kbd> in your art program (once the eraser is lifted the E is pressed again), as if it were a pencil.

ThatOneCalculator, 2022-09-10:
From `/usr/lib/linux/input-event-codes.h`:

```
#define BTN_MISC		0x100
#define BTN_0			0x100
#define BTN_1			0x101
#define BTN_2			0x102
#define BTN_3			0x103
#define BTN_4			0x104
#define BTN_5			0x105
#define BTN_6			0x106
#define BTN_7			0x107
#define BTN_8			0x108
#define BTN_9			0x109

// ...

#define BTN_DIGI		0x140
#define BTN_TOOL_PEN		0x140
#define BTN_TOOL_RUBBER		0x141
#define BTN_TOOL_BRUSH		0x142
#define BTN_TOOL_PENCIL		0x143
#define BTN_TOOL_AIRBRUSH	0x144
#define BTN_TOOL_FINGER		0x145
#define BTN_TOOL_MOUSE		0x146
#define BTN_TOOL_LENS		0x147
#define BTN_TOOL_QUINTTAP	0x148	/* Five fingers on trackpad */
#define BTN_STYLUS3		0x149
#define BTN_TOUCH		0x14a
#define BTN_STYLUS		0x14b
#define BTN_STYLUS2		0x14c
#define BTN_TOOL_DOUBLETAP	0x14d
#define BTN_TOOL_TRIPLETAP	0x14e
#define BTN_TOOL_QUADTAP	0x14f	/* Four fingers on trackpad */
```

I wonder which of these it'd be :thinking:

kurikaesu, 2022-09-11:
Probably BTN_STYLUS3 but no idea if that is what that event is actually used for. The question of course is if it also sends through pressure like on the wacom styluses?

ThatOneCalculator, 2022-09-11:
It does

kurikaesu, 2022-09-11:
I was wrong. It is BTN_TOOL_RUBBER. Used by the wacom drivers:

https://github.com/torvalds/linux/blob/master/drivers/hid/wacom_wac.c#L134-L135

PR #64 (2022-11-16): Add initial Eraser support to xp_pen_unified_device.cpp
I took inspiration from #62 and tried to rework it to only press a custom key when entering proximity. I also added the ability to add keybinds to the pen tip entering proximity so you can switch back and forth between tools when alternating ends of the pen.

[kurikaesu/userspace-tablet-driver-gui#14](https://github.com/kurikaesu/userspace-tablet-driver-gui/pull/14) adds support for binding keybinds to eraser proximity entry and could be easily extended to add support for pen proximity as well.

This should close #60.
closed 2022-11-16 by kurikaesu