kurikaesu/userspace-tablet-driver-daemon #66: XP-Pen Deco mini7 works only partially

CodedSakura, 2022-12-23:
_while installing this repo/package did not seem to impact anything, this repo does seem to be the best place to report this_

I have a XP-Pen Deco mini7 (`28bd:0928`) and while by default it works, it's not seen by `xsetwacom`. Making a `/etc/X11/xorg.conf.d/10-tablet.conf` with the following code seemed to fix that, but `xsetwacom list` does not show a `pad` device and tablet buttons don't work at all.
```
Section "InputClass"
  Identifier "XP-Pen Deco mini7"
  MatchUSBID "28bd:0928"
  MatchDevicePath "/dev/input/event*"
#  Driver "libinput"
  Driver "wacom"
EndSection
```

if the driver is set to `"libinput"` like in this repo's `/config/usr/share/X11/xorg.conf.d/60-xp-pen.conf` file, the tablet acts the same as before adding the file.

system: 
linux 6.1.1.arch1-1
libwacom 2.5.0-1
xf86-input-wacom 1.1.0-1

any help/tips to to get the tablet to show up in `xsetwacom` with a `pad` device?

kurikaesu, 2022-12-23:
Hello, this driver does not work under libwacom as we don't expose the tablet's details to the wacom driver as it expects it to be.

Is there anything you need specifically from the use of `xsetwacom` ?

If you absolutely need to use the wacom driver, then you should instead see if the digimend drivers has a pull request that will support your device like this: https://github.com/DIGImend/digimend-kernel-drivers/issues/487

CodedSakura, 2022-12-24:
ah, i guess i was confused then.
while i don't really care if i can use `xsetwacom` or not, having does enable quite a few integrations & useful GUI apps.

for now i just want to be able to rebind the buttons. how do i set up this daemon then and rebind the buttons?
i've already tried reverting the `10-tablet.conf` driver to `"libinput"`, but the tablet does not show up in your GUI util..

CodedSakura, 2022-12-24:
alright, i got it working!
turns out i had to run the ~~nowhere~~ subtly mentioned `userspace_tablet_driver_daemon` command. i would appreciate if better setup instructions were added :p

closed 2022-12-24 by CodedSakura