kurikaesu/userspace-tablet-driver-daemon #68: How to get "userspace-tablet-driver-daemon" work on FreeBSD 13.1?

rajhlinux, 2023-01-05:
Alright, I have built it on FreeBSD 13.1

I ran the command:
`userspace_tablet_driver_daemon`

And this is the only output I get:
```
Listening on socket /home/user/.local/var/run/userspace_tablet_driver_daemon.sock
xp_pen_handler initialized
huion_handler initialized
Saved updated configuration file
```

Nothing else happens...

I have a XP-Pen Star G640 tablet.

Here are my PC info:
OS: FreeBSD 13.1-RELEASE-p5 amd64 
Uptime: 1 hour, 12 mins 
Packages: 1637 (pkg) 
Shell: sh 
Resolution: 3840x2160 
DE: Plasma 5.24.7 
WM: KWin 
Theme: [Plasma], Breeze [GTK2/3] 
Icons: [Plasma], breeze-dark [GTK2/3] 
Terminal: konsole 
CPU: AMD FX-8350 (8) @ 3.991GHz 
GPU: Ellesmere [Radeon RX 470/480/570/570X/580/580X/590] 
Memory: 12385MiB / 32684MiB

Thanks.

kurikaesu, 2023-01-05:
If the driver does not detect your device it is likely because it does not use the alternate unlock bit that is used by the current generation of XP-Pen devices.

In this case, the device will need to have drivers written specifically targeting it. If you know or would like to try getting libinput/termcap to give you USB packet dumps you can try and implement it yourself.

Apart from that the only alternatives is someone else who owns the tablet takes on the implementation or I go and buy the tablet myself and do it myself.

rajhlinux, 2023-01-06:
Thank You for your reply.

Wow that is interesting. 
It makes sense why so many numerous different implementations did not work.

I will try to give many information and see if we can get this to work:

When I do: `xsetwacom list devices`
The return is simply blank.

Here is what `xinput list` shows:

```
⎡ Virtual core pointer                          id=2    [master pointer  (3)]

⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]

⎜   ↳ System mouse                              id=6    [slave  pointer  (2)]

⎜   ↳ Logitech Gaming Mouse G502                id=12   [slave  pointer  (2)]

⎜   ↳ UGTABLET 6 inch PenTablet                 id=13   [slave  pointer  (2)]

⎜   ↳ UGTABLET 6 inch PenTablet                 id=14   [slave  pointer  (2)]

⎜   ↳ BY Tech Usb Gaming Keyboard               id=15   [slave  pointer  (2)]

⎣ Virtual core keyboard                         id=3    [master keyboard (2)]

    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]

    ↳ System keyboard multiplexer               id=7    [slave  keyboard (3)]

    ↳ Power Button                              id=8    [slave  keyboard (3)]

    ↳ AT keyboard                               id=9    [slave  keyboard (3)]

    ↳ Logitech Gaming Mouse G502                id=10   [slave  keyboard (3)]

    ↳ BY Tech Usb Gaming Keyboard               id=11   [slave  keyboard (3)]
```

Here is what `dmesg | grep -i PenTablet` outputs:

```
ugen0.2: <UGTABLET 6 inch PenTablet> at usbus0

ums2: <UGTABLET 6 inch PenTablet, class 0/0, rev 1.10/0.00, addr 1> on usbus0

ums3: <UGTABLET 6 inch PenTablet, class 0/0, rev 1.10/0.00, addr 1> on usbus0

uhid0: <UGTABLET 6 inch PenTablet, class 0/0, rev 1.10/0.00, addr 1> on usbus0
```

Here is what `xorg.log` shows:

```
[    13.485] (II) config/udev: Adding input device UGTABLET 6 inch PenTablet (/dev/input/event7)

[    13.485] (**) UGTABLET 6 inch PenTablet: Applying InputClass "evdev pointer catchall"

[    13.485] (**) UGTABLET 6 inch PenTablet: Applying InputClass "libinput pointer catchall"

[    13.485] (II) Using input driver 'libinput' for 'UGTABLET 6 inch PenTablet'

[    13.485] (**) UGTABLET 6 inch PenTablet: always reports core events

[    13.486] (**) Option "Device" "/dev/input/event7"

[    13.486] (**) Option "_source" "server/udev"

[    13.487] (II) event7  - UGTABLET 6 inch PenTablet, class 0/0, rev 1.10/0.00, addr 1: is tagged by udev as: Mouse

[    13.488] (II) event7  - UGTABLET 6 inch PenTablet, class 0/0, rev 1.10/0.00, addr 1: device is a pointer

[    13.490] (II) event7  - UGTABLET 6 inch PenTablet, class 0/0, rev 1.10/0.00, addr 1: device removed

[    13.490] (**) Option "config_info" "udev:/dev/input/event7"

[    13.490] (II) XINPUT: Adding extended input device "UGTABLET 6 inch PenTablet" (type: MOUSE, id 13)

[    13.490] (**) Option "AccelerationScheme" "none"

[    13.490] (**) UGTABLET 6 inch PenTablet: (accel) selected scheme none/0

[    13.490] (**) UGTABLET 6 inch PenTablet: (accel) acceleration factor: 2.000

[    13.490] (**) UGTABLET 6 inch PenTablet: (accel) acceleration threshold: 4

[    13.492] (II) event7  - UGTABLET 6 inch PenTablet, class 0/0, rev 1.10/0.00, addr 1: is tagged by udev as: Mouse

[    13.493] (II) event7  - UGTABLET 6 inch PenTablet, class 0/0, rev 1.10/0.00, addr 1: device is a pointer

[    13.495] (II) config/udev: Adding input device UGTABLET 6 inch PenTablet (/dev/input/event8)

[    13.495] (**) UGTABLET 6 inch PenTablet: Applying InputClass "evdev pointer catchall"

[    13.495] (**) UGTABLET 6 inch PenTablet: Applying InputClass "libinput pointer catchall"

[    13.495] (II) Using input driver 'libinput' for 'UGTABLET 6 inch PenTablet'

[    13.495] (**) UGTABLET 6 inch PenTablet: always reports core events

[    13.495] (**) Option "Device" "/dev/input/event8"

[    13.495] (**) Option "_source" "server/udev"

[    13.497] (II) event8  - UGTABLET 6 inch PenTablet, class 0/0, rev 1.10/0.00, addr 1: is tagged by udev as: Mouse

[    13.498] (II) event8  - UGTABLET 6 inch PenTablet, class 0/0, rev 1.10/0.00, addr 1: device is a pointer

[    13.500] (II) event8  - UGTABLET 6 inch PenTablet, class 0/0, rev 1.10/0.00, addr 1: device removed

[    13.500] (**) Option "config_info" "udev:/dev/input/event8"

[    13.500] (II) XINPUT: Adding extended input device "UGTABLET 6 inch PenTablet" (type: MOUSE, id 14)

[    13.500] (**) Option "AccelerationScheme" "none"

[    13.500] (**) UGTABLET 6 inch PenTablet: (accel) selected scheme none/0

[    13.500] (**) UGTABLET 6 inch PenTablet: (accel) acceleration factor: 2.000

[    13.500] (**) UGTABLET 6 inch PenTablet: (accel) acceleration threshold: 4

[    13.502] (II) event8  - UGTABLET 6 inch PenTablet, class 0/0, rev 1.10/0.00, addr 1: is tagged by udev as: Mouse

[    13.503] (II) event8  - UGTABLET 6 inch PenTablet, class 0/0, rev 1.10/0.00, addr 1: device is a pointer
```

Here is what `inputplug -v -d -0 -c XISlavePointer` shows:
(inputplug is Graphical tool for configuring pointing devices, testing xinput events)

```
"XISlavePointer" "XISlaveAdded" "13" "XISlavePointer" "UGTABLET 6 inch PenTablet"
Command failed: No such file or directory (os error 2)
"XISlavePointer" "XIDeviceEnabled" "13" "XISlavePointer" "UGTABLET 6 inch PenTablet"
Command failed: No such file or directory (os error 2)
"XISlavePointer" "XISlaveAdded" "14" "XISlavePointer" "UGTABLET 6 inch PenTablet"
Command failed: No such file or directory (os error 2)
"XISlavePointer" "XIDeviceEnabled" "14" "XISlavePointer" "UGTABLET 6 inch PenTablet"
Command failed: No such file or directory (os error 2)
```

Thanks.

kurikaesu, 2023-01-16:
On the positive side, it looks like libinput is working correctly since it is assuming control over the device.

I've gone ahead and bought the tablet. I'll see if I can implement the device when I receive it.

kurikaesu, 2023-02-11:
Commit https://github.com/kurikaesu/userspace-tablet-driver-daemon/commit/954565d8d3ff11001d0d0297472e046514bf0367 should add support for the G640. Please let me know if it does not work for you.

kurikaesu, 2025-09-15:
Closing this out since there hasn't been any response in years.

closed 2025-09-15 by kurikaesu