kurikaesu/userspace-tablet-driver-daemon #69: the monitor remapping problem in startup

sm633, 2023-01-05:
Hi!
I am sm633.
I was very surprised to find this open source driver.
Thank you for making this driver!

I'm using XP-PEN Artist Pro 16 on Arch Linux.
I'm using a multi-display setup with another display connected via DVI to the NVIDIA graphics card.

In GNOME, the Tablet was firmly mapped to the main screen, but in the i3 Window Manager, it is mapped across two displays.

After starting the i3 Window Manager, run xinput map-to-output "HANVON UGEE Artist Pro 16 Pen (0)" "HDMI-0"
I can map it correctly, but when I run it with XDG Autostart, the mapping fails.

Running "xrandr --listmonitors" resulted in

Monitors: 2
  0: +*DVI-D-0 1920/476x1080/268+0+0 DVI-D-0
  1: +HDMI-0 1920/344x1080/193+1920+0 HDMI-0

As a result of running "xinput",

⎡ Virtual core pointer id=2 [master pointer (3)]
⎜ ↳ Virtual core XTEST pointer id=4 [slave pointer (2)]
⎜ ↳ Gtech wireless dongle Mouse id=9 [slave pointer (2)]
⎜ ↳ Gtech wireless dongle Consumer Control id=10 [slave pointer (2)]
⎜ ↳ ARCHISS M91JP Mouse id=13 [slave pointer (2)]
⎜ ↳ ARCHISS M91JP Consumer Control id=15 [slave pointer (2)]
⎜ ↳ HANNON UGEE Artist Pro 16 Mouse id=17 [slave pointer (2)]
⎣ Virtual core keyboard id=3 [master keyboard (2)]
     ↳ Virtual core XTEST keyboard id=5 [slave keyboard (3)]
     ↳ Power Button id=6 [slave keyboard (3)]
     ↳ Power Button id=7 [slave keyboard (3)]
     ↳ Gtech wireless dongle id=8 [slave keyboard (3)]
     ↳ Gtech wireless dongle System Control id=11 [slave keyboard (3)]
     ↳ ARCHISS M91JP id=12 [slave keyboard (3)]
     ↳ ARCHISS M91JP System Control id=14 [slave keyboard (3)]
     ↳ ARCHISS M91JP id=16 [slave keyboard (3)]
     ↳ Hanvon UGEE Artist Pro 16 Keyboard id=18 [slave keyboard (3)]
     ↳ Hanvon UGEE Artist Pro 16 id=19 [slave keyboard (3)]
     ↳ Eee PC WMI hotkeys id=20 [slave keyboard (3)]
     ↳ Gtech wireless dongle Consumer Control id=21 [slave keyboard (3)]
     ↳ ARCHISS M91JP Consumer Control id=22 [slave keyboard (3)]

"HANVON UGEE Artist Pro 16 Pen (0)" is not recognized by xinput when XDG Autostart is started.

After the i3 Window Manager has started
"HANVON UGEE Artist Pro 16 Pen (0)"
is recognized.

Could you please tell me how to solve this problem?

kurikaesu, 2023-01-16:
I'm not sure what the start up order is for XDG but this is probably beyond the scope of the support I can provide for this as I don't use the i3 window manager and don't currently have a Linux machine to do testing with.

I'll have to leave this issue open to allow someone else with i3 experience to chime in if possible.

minneyar, 2023-03-14:
I'm using GNOME, but I think I'm having a similar issue.  I have an XP-Pen Artist 13.3 Pro that I'm using as a secondary display.  I can run `xinput map-to-output "XP-Pen Artist 13.3 Pro Pen (0)" HDMI-0` to map the pen to the display, but the problem is that the `XP-Pen Artist 13.3 Pro Pen (0)` device does not appear immediately when the tablet is plugged in; it only appears after the first time the tablet has received input from the pen.

When starting everything up from scratch, my sequence of events looks like:

1. Boot everything up, execute `userspace_tablet_driver_daemon`.
2. Plug in my tablet.
3. `userspace_tablet_driver_daemon` recognizes the tablet and initializes it.  `kern.log` contains quite a bit of output as it recognizes devices, and `udevadm monitor` also prints a lot of output that I won't paste here because it's fairly long.

At this point, moving my stylus over the tablet maps its input across all of my monitors, which is obviously wrong.  Running `xinput map-to-output "XP-Pen Artist 13.3 Pro Pen (0)" HDMI-0` prints `unable to find device 'XP-Pen Artist 13.3 Pro Pen (0)'`, and I can verify that the device doesn't appear in `xinput list`.

But as soon as I wave my pen over the tablet, `syslog` prints:
```
Mar 14 00:26:34 xenogears /usr/libexec/gdm-x-session[3001]: (II) libinput: XP-Pen Artist 13.3 Pro: needs a virtual subdevice
Mar 14 00:26:34 xenogears /usr/libexec/gdm-x-session[3001]: (**) XP-Pen Artist 13.3 Pro Pen (0): Applying InputClass "libinput tablet catchall"
Mar 14 00:26:34 xenogears /usr/libexec/gdm-x-session[3001]: (II) Using input driver 'libinput' for 'XP-Pen Artist 13.3 Pro Pen (0)'
Mar 14 00:26:34 xenogears /usr/libexec/gdm-x-session[3001]: (II) systemd-logind: returning pre-existing fd for /dev/input/event20 13:84
Mar 14 00:26:34 xenogears /usr/libexec/gdm-x-session[3001]: (**) XP-Pen Artist 13.3 Pro Pen (0): always reports core events
Mar 14 00:26:34 xenogears /usr/libexec/gdm-x-session[3001]: (**) Option "Device" "/dev/input/event20"
Mar 14 00:26:34 xenogears /usr/libexec/gdm-x-session[3001]: (II) libinput: XP-Pen Artist 13.3 Pro Pen (0): is a virtual subdevice
Mar 14 00:26:34 xenogears /usr/libexec/gdm-x-session[3001]: (**) Option "config_info" "udev:/sys/devices/virtual/input/input236/event20"
Mar 14 00:26:34 xenogears /usr/libexec/gdm-x-session[3001]: (II) XINPUT: Adding extended input device "XP-Pen Artist 13.3 Pro Pen (0)" (type: STYLUS, id 22)
Mar 14 00:26:34 xenogears /usr/libexec/gdm-x-session[3001]: (**) Option "AccelerationScheme" "none"
Mar 14 00:26:34 xenogears /usr/libexec/gdm-x-session[3001]: (**) XP-Pen Artist 13.3 Pro Pen (0): (accel) selected scheme none/0
Mar 14 00:26:34 xenogears /usr/libexec/gdm-x-session[3001]: (**) XP-Pen Artist 13.3 Pro Pen (0): (accel) acceleration factor: 2.000
Mar 14 00:26:34 xenogears /usr/libexec/gdm-x-session[3001]: (**) XP-Pen Artist 13.3 Pro Pen (0): (accel) acceleration threshold: 4
Mar 14 00:26:34 xenogears gsd-wacom[3307]: file ../libwacom/libwacom.c: line 1005 (libwacom_set_default_match): should not be reached
```

After this, I can run `xinput map-to-output "XP-Pen Artist 13.3 Pro Pen (0)" HDMI-0` to map the input appropriately -- but `udevadm monitor` doesn't print anything, so I don't know if it's possible to create a udev rule that responds to it.  Is there any other way to automate setting up the mapping as soon as the stylus is detected?

kurikaesu, 2023-03-14:
Have you tried something like what is described here?
https://unix.stackexchange.com/questions/65891/how-to-execute-a-shellscript-when-i-plug-in-a-usb-device

You'll have to adapt the answer to detecting your display instead.
There may also be an issue that the X display variables are not available and will need to be manually set in the udev rule like in this question: https://askubuntu.com/questions/1269127/xrandr-fails-when-run-from-udev which refers to this post: https://frdmtoplay.com/i3-udev-xrandr-hotplugging-output-switching/

minneyar, 2023-03-15:
That doesn't seem to work for me.  I can execute a script when I plug in my tablet, but that isn't actually useful, because the virtual pointer device does not appear at the same time the USB device is connected.  The virtual pointer doesn't appear in `xinput list` until after I actually move the stylus over my tablet, but that does not trigger any udev events.

Immediately after plugging in my tablet, the output of `xinput list` looks like this (omitting unrelated devices):
```
$ xinput list 
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ XP-Pen Artist 13.3 Pro                  	id=19	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ XP-Pen Artist 13.3 Pro                  	id=20	[slave  keyboard (3)]
    ↳ XP-Pen Artist 13.3 Pro                  	id=21	[slave  keyboard (3)]
```

Running `xinput map-to-output 19 HDMI-0` returns without error, but when I try to actually use my stylus, the mapping is obviously wrong.

After I bring my stylus over the screen, another device appears in the list:
```
$ xinput list 
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ XP-Pen Artist 13.3 Pro                  	id=19	[slave  pointer  (2)]
⎜   ↳ XP-Pen Artist 13.3 Pro Pen (0)          	id=22	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ XP-Pen Artist 13.3 Pro                  	id=20	[slave  keyboard (3)]
    ↳ XP-Pen Artist 13.3 Pro                  	id=21	[slave  keyboard (3)]
```

Running `xinput map-to-output 22 HDMI-0` set the mapping correctly, but I can't find any way to run that command when that device appears since it does not produce a udev event.

kurikaesu, 2023-03-15:
I'll have to try it out with my own hardware to see what the sequence of events looks like. Weird that the monitor doesn't exist in xinput until the stylus is used. Would the same thing happen if a regular monitor was attached to the computer?

minneyar, 2023-03-15:
The monitor exists immediately; I can see it right away with:

```
$ xrandr --listmonitors
Monitors: 3
 0: +*DP-0 2560/597x1440/336+1920+598  DP-0
 1: +DP-2 1440/597x2560/336+4480+0  DP-2
 2: +HDMI-0 1920/294x1080/165+0+74  HDMI-0
```

The tablet is HDMI-0; DP-0 and DP-2 are my other monitors.  It seems like it's just the xinput virtual pointer device representing the stylus that doesn't exist.

minneyar, 2023-03-22:
Just thought I'd comment that I figured out a workaround for this issue.

I figured out that running `xinput map-to-output` sets a coordinate transformation matrix on the pointer device as a property.  This property is lost when the device is removed, but it's possible to manually override the default transformation matrix in Xorg's configuration.

First, I plugged in my tablet, touched my stylus to the screen to make the virtual pointer device appear, and then ran this to set it:

```
xinput map-to-output "XP-Pen Artist 13.3 Pro Pen (0)" HDMI-0
```

Now it's possible to see the transformation matrix by using the `list-props` command:

```
$ xinput list-props "XP-Pen Artist 13.3 Pro Pen (0)"
Device 'XP-Pen Artist 13.3 Pro Pen (0)':
	Device Enabled (155):	1
	Coordinate Transformation Matrix (157):	0.324324, 0.000000, 0.000000, 0.000000, 0.421875, 0.028906, 0.000000, 0.000000, 1.000000
	libinput Send Events Modes Available (275):	1, 0
	libinput Send Events Mode Enabled (276):	0, 0
	libinput Send Events Mode Enabled Default (277):	0, 0
	Device Node (278):	"/dev/input/event3"
	Device Product ID (279):	10429, 63787
	libinput Tablet Tool Pressurecurve (721):	0.000000, 0.000000, 0.000000, 0.000000, 1.000000, 1.000000, 1.000000, 1.000000
	libinput Tablet Tool Area Ratio (722):	0, 0
```

Next, I edited `/usr/share/X11/xorg.conf.d/60-xp-pen.conf`.  It has a separate config block for every input device, so I found the one for "XP-Pen Artist 13.3 Pro" and set added the TransformationMatrix like this:

```
Section "InputClass"
  Identifier "XP-Pen Artist 13.3 Pro"
  MatchUSBID "28bd:f92b"
  MatchDevicePath "/dev/input/event*"
  Option "TransformationMatrix" "0.324324 0.000000 0.000000 0.000000 0.421875 0.028906 0.000000 0.000000 1.000000"
  Driver "libinput"
EndSection
```

Note that the matrix in the output of `xinput list-props` has commas, but the one in `60-xp-pen.conf` does not.

After restarting, now the mapping for my stylus is automatically set every time I connect it.

The only catch is that these values will be specific to the monitor arrangement, so it has to be manually updated if that's changed, which is a little annoying, but not a big problem.