kurikaesu/userspace-tablet-driver-daemon #70: XP-Pen Artist 22E Pro not recognized

Mnc2g, 2023-01-15:
Hi, your piece of software is great, however my XP-Pen Artist 22E Pro is not recognized, I get the following error when I run
`userspace_tablet_driver_daemon &` (which, I think, is the right way to run it since it's the only way I can make it work):

```
xp_pen_handler initialized
huion_handler initialized
Saved updated configuration file
Setup completed on interface 0
Setup completed on interface 1
Device: Generic XP-Pen Device - Probed maxWidth: (47663) maxHeight: (26809) resolution: (2540)
Attached to interface 2
Sending init key on endpont 3
Setup completed on interface 2
Set up config for device 36875: (Generic XP-Pen Device)
Unknown product 36875
```
Do you have any ideas, or is it the way it should work?

Many thanks,
Silvio

kurikaesu, 2023-01-15:
Hello, the device is not officially recognised by the driver but it looks like it is still at least attempting to get it running.

Does the tablet work, including the stylus buttons?

If it does work, it would be a matter of adding a new set of classes for the Artist 22E Pro to provide "official" support

Mnc2g, 2023-01-16:
Hi, buttons on stylus are working, the ones on the screen frame are also
working.
Could you please tell me how to customize them?
Many thanks.

kurikaesu, 2023-01-16:
You can take this PR as a base for what needs to be done:

https://github.com/kurikaesu/userspace-tablet-driver-daemon/pull/65/commits/7474e60bc936451d4481052e2123de7ceee4e643

Once the driver is able to properly recognise your tablet, you would then need to update the GUI to also understand the new device like this:

https://github.com/kurikaesu/userspace-tablet-driver-gui/pull/15/commits/024aeb6314df3ea0db2bf47243a567ea79152aa4

Until the driver is no longer using the generic driver and the GUI handles it properly, the only way to make any changes to bindings would be to hard-code the new bindings you want to the generic driver in code.

I would advise against doing that and instead making similar changes to the above PRs. This way, others will also be able to enjoy proper support.

Send me a PR when you are ready!

kurikaesu, 2023-01-16:
If you aren't able to make the changes, I'll see if I can make them later today when I get some time.

Mnc2g, 2023-01-17:
Thank you,
Unfortunately, I can't make the changes, moreover I wasn't able to install
the GUI, and therefore I asked for a configuration file.

Many thanks.Message ID:
***@***.***>

>

coffseducation, 2023-02-02:
This affects me too:

userspace_tablet_driver_daemon &
[2] 6783
anarcho@Vampa:~$ Listening on socket /home/anarcho/.local/var/run/userspace_tablet_driver_daemon.sock
xp_pen_handler initialized
huion_handler initialized
Handling XP-Pen Artist 22E Pro
Could not claim interface 0 retcode: -6 errno: 16
Could not claim device on attempt 0. Detaching and then waiting
Could not claim interface 0 retcode: -6 errno: 16
Could not claim device on attempt 1. Detaching and then waiting
Could not claim interface 0 retcode: -6 errno: 16
Could not claim device on attempt 2. Detaching and then waiting
Could not claim interface 0 retcode: -6 errno: 16
Could not claim device on attempt 3. Detaching and then waiting
Could not claim interface 0 retcode: -6 errno: 16
Could not claim device on attempt 4. Detaching and then waiting
Giving up

coffseducation, 2023-02-02:
It is actually working now:

userspace_tablet_driver_daemon &
[1] 8878
anarcho@Vampa:~$ Listening on socket /home/anarcho/.local/var/run/userspace_tablet_driver_daemon.sock
xp_pen_handler initialized
huion_handler initialized
Handling XP-Pen Artist 22E Pro
Setup completed on interface 0
Setup completed on interface 1
Device: XP-Pen Artist 22E Pro - Probed maxWidth: (47663) maxHeight: (26809) resolution: (2540)
Attached to interface 2
Sending init key on endpont 3
Setup completed on interface 2
Set up config for device 2315: (XP-Pen Artist 22E Pro)

The only problem have is alignment of the pen - but I cannot launch the gui to set this. I put in a bug report separately though.

kurikaesu, 2023-02-02:
If the problem is that your pen and display need calibration as the cursor is offset a little, then you will have to use the X11 calibration tools if you are still using X11/xorg.

If you are on wayland then there is no current way to calibrate the tablets unless KDE, Gnome or other window managers have provided a way to do so.

coffseducation, 2023-02-06:
> X11 calibration tools if you are still using X11/xorg.

Thanks! I did not know that these existed! I will give them a try! 🙏

kurikaesu, 2023-02-11:
Closing this. Please re-open if you need more help!

closed 2023-02-11 by kurikaesu