kurikaesu/userspace-tablet-driver-daemon #71: Package openjdk-16-jre is not installed.

coffseducation, 2023-02-02:
Hello,

I am trying to install userspace-tablet-driver-utility.deb but am getting a dependency error for openjdk-16-jre.
Package openjdk-16-jre is not in the 20.10 repo, so I installed Package openjdk-20-jre.
It still gives me the same error.
How can I install userspace-tablet-driver-utility.deb and satisfy dependencies?
Thanks!

sudo dpkg -i userspace-tablet-driver-utility.deb 
[sudo] password for anarcho: 
(Reading database ... 653716 files and directories currently installed.)
Preparing to unpack userspace-tablet-driver-utility.deb ...
Unpacking xp-pen-userland-util (0.1.0) over (0.1.0) ...
dpkg: dependency problems prevent configuration of xp-pen-userland-util:
 xp-pen-userland-util depends on openjdk-16-jre; however:
  Package openjdk-16-jre is not installed.

dpkg: error processing package xp-pen-userland-util (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 xp-pen-userland-util

Operating System: Kubuntu 22.10
KDE Plasma Version: 5.26.5
KDE Frameworks Version: 5.101.0
Qt Version: 5.15.6
Kernel Version: 5.19.0-1015-lowlatency (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-6700 CPU @ 3.40GHz
Memory: 31.2 GiB of RAM
Graphics Processor: Mesa Intel® HD Graphics 530
Manufacturer: Dell Inc.
Product Name: Precision 5720 AIO

Tablet: Xp-pen 22E Pro

kurikaesu, 2023-02-02:
Please try using OpenJDK 11. I haven't updated the code to work correctly with OpenJDK 14+ so unless someone does the required modification to do so, it will not work.

coffseducation, 2023-02-06:
Thanks for the reply!

Kubuntu was on OpenJDK 11
This gave me:
xp-pen-userland-util depends on openjdk-16-jre; however:
which is why I tried upgrading OpenJDK. 🤷🏻‍♂

fitzsim, 2026-01-04:
I ended up rebuilding the utility from source code using `temurin-11-jdk`.  The `tornadofx` `2.0.0` snapshot didn't resolve so I had to rebuild it from source too.  When I built the deb I changed the dependency to `temurin-11-jre`, which got the GUI program building/installing/working.  I can put together a pull request with all this if there is interest.

kurikaesu, 2026-01-04:
Temurin is a different distribution of OpenJDK? I'd be interested to see a PR on that and try it out myself