kurikaesu/userspace-tablet-driver-daemon #75: The driver works for Artist Pro 16 (Gen2)

Vineyo, 2023-07-17:
Info for the device:
Vendor ID | 0x28bd | (XP-Pen)
Product ID | 0x095b | 
Link to the product: https://www.xp-pen.com.cn/product/888.html
Though the GUI doesn't recognize it.

Deevad, 2023-10-31:
Hey @Vineyo , can you tell what is working on your side? I [have the device](https://framapiaf.org/@davidrevoy/111318263108091568) since a couple days. On X11, it works almost fully on the generic uc-logic/ugee kernel driver; as soon a Xorg rule is done. Eg. `/usr/share/X11/xorg.conf.d/60-xppen.conf`:

```
Section "InputClass"
  Identifier "XP-Pen Artist Pro 16 (Gen2) Frame"
  MatchUSBID "28bd:095b"
  MatchIsKeyboard "on"
  MatchDevicePath "/dev/input/event*"
  Driver "libinput"
EndSection

Section "InputClass"
  Identifier "XP-Pen Artist Pro 16 (Gen2) Tablet"
  MatchIsTablet "on"
  Driver "wacom"
  MatchUSBID "28bd:095b"
  MatchDevicePath "/dev/input/event*"
EndSection
```
Then reboot, then it appears handled by `xsetwacom` tool,  the tablet eraser and stylus appears with a `$xsetwacom list`:
 
```
UGTABLET Artist Pro 16 (Gen2) stylus    id: 13  type: STYLUS    
UGTABLET Artist Pro 16 (Gen2) eraser    id: 17  type: ERASER 
```
After that, the stylus is fully configurable (almost) with xsetwacom commands, and some are exposed with kcm-tablet (the GUI of Kde/Plasma on X11). On Wayland, it has a minimal support, just enough to change monitor.

The 'almost part' is about the second button on stylus: the top one. It reacts on my side as a hardcoded Eraser button. As soon as press it, the stylus tip become an Eraser device, and exit the mode on release. This makes this button impossible to customise via xsetwacom even if the button 3 exists. 

I tested this repo, I even tried to make a definition for the tablet (based on the files of the Pro 12: 
[artist_16_pro_gen2.zip](https://github.com/kurikaesu/userspace-tablet-driver-daemon/files/13218202/artist_16_pro_gen2.zip) ) , I can compile, but as soon as I run a final `userspace_tablet_driver_daemon` to check (even after a reboot), I still don't see the tablet. 

So, I'm curious how you got the driver here to work with this model @Vineyo .

Vineyo, 2023-11-01:
I'm sorry that I used this driver as a quick fix because I got the tablet before the official driver was released. I didn't test the buttons. And I'm using the proprietary driver.

Deevad, 2023-11-01:
Good to know and no problem! Thank you for the feedback. 

I posted today on my blog a post about my issue https://www.davidrevoy.com/article995/how-a-kernel-update-broke-my-stylus-need-help , the comment section was active and helpful and redirected me to contact the right poeple. Finger crossed for a good Free/Libre and Open Source support.

kurikaesu, 2023-11-01:
Would love to help and possibly provide a kernel patch but I am currently traveling. Would also be interested in knowing how this kernel change possibly affected my driver.